SWC-Advanced-Microscopy/BakingTray #184: Log attempts to regain modelock and try a bit harder.

raacampbell, 2019-05-01:
BakingTray could be a lot more robust to failure of the laser. It should pause the acquisition and try harder to regain modelocking.

raacampbell, 2019-05-01:
Further, sections where the laser failed do not have the COMPLETED file.

raacampbell, 2019-05-01:
FYI @ablot 
Modelock failure is tested after the tile scan. If the laser is not modelocked once the tile scan is complete, COMPLETED file is not created and the `bake` method ends prematurely (although it will run its "cleanup" function). Therefore when the acquisition is resumed the section with missing tiles will be re-imaged. The sections associated with modelock failures are logged in the main log file in the sample root directory.

raacampbell, 2019-05-01:
The messages are, oddly, that the laser has turned off. Not that it's lost modelock. So to be sure I add code to explicitly check this. Then add code to try turning it back on again.

raacampbell, 2019-05-02:
I finish acquisition by selecting "stop after section" and I get odd errors that make no sense:
[image]
Unless it's just chance that the laser turns off at this point, this is most odd.

raacampbell, 2019-05-02:
To do:
* [ ] Monitor tower temp and report if above 25 degrees. 
* [ ] Laser returns status codes indicating previous errors. Set up a system for reporting those. Either when there is a failure or at some other time.

ablot, 2019-05-09:
Thanks.
I assumed that the reimaged section would be the next section (i.e. if the laser failed in the middle of section 70, I would have a folder `section_70` with half of the tiles being black and and a `section_71` folder with the same section imaged when I restarted the `bake`). 
Is that right?

raacampbell, 2019-05-09:
I think that is correct. But the stitching doesn't know about that.

Rob

raacampbell, 2019-06-07:
At least based on the Slack messages, it seems that laser modelock failures are now being caught and fixed on many occasions (50ea51ceee0a0c0d8ad9f3256ec8153dbb91f1a4), although I don't know for sure that the laser had lost modelock. Maybe it's lying. 

The repeated failures we saw previously were due to the laser getting too warm. In one instance the light shroud curtain was blocking the intake vents one the chiller. This is no longer possible.

closed 2019-06-07 by raacampbell