SWC-Advanced-Microscopy/BakingTray #314: Lauching acquisition with Average Factor not set to 1

HugoB, 2020-11-04:
Hello Rob, 

I am trying to launch an acquisition while averaging each frame 5 or 10 times. I tried with big stacks of around 70 z plans but the cache memory exploded and Matlab crashed (64 Gb full).
So I tried with fewer plans (6) but know the matlab console is spamming : "ERROR in everyNEvent callback of Task object: Index exceeds array bounds."

Is there a workaround to do an acquisition while averaging the frames ?

raacampbell, 2020-11-04:
For linear scanning do not average frames. Change your scan settings so that the line period and dwell time increase. e.g. increase the number of samples per pixel.

HugoB, 2020-11-04:
Ok !
Thank you.

raacampbell, 2020-11-12:
This is now mentioned on this documentation page: https://bakingtray.mouse.vision/users/choosing-resolution

closed 2020-11-12 by raacampbell