SWC-Advanced-Microscopy/BakingTray #328: Missing lines in the acqLog

HugoB, 2020-12-09:
Hello Rob, 

I just merged my fork with the current master and did an acquisition, which worked fine except some small bugs. 
One of them is the absence the 2 lines in the acqLog : 
I had the log beginning like this : 
"
2020/12/08 11:31:12 -- STARTING NEW ACQUISITION
Writing to volume Z: which has 60453/90872 GB free
Using laser: Chameleon MPX, Version, Serial Number, Device Type: APE GmbH, Chameleon OPO, S05221, V 0.5.3376, SW4.4.19.6
Acquiring with: ScanImage v5.6.1 on MATLAB 9.5.0.944444 (R2018b)
Using BakingTray version 210015c6cc6858f13f8e51716bf057d6aedd38ca from branch master
Setting laser watchdog timer to 2400 seconds
...
"
instead of something like this :
"
Writing to volume Z: which has 59985/90872 GB free
2020/10/30 13:57:48 -- STARTING NEW ACQUISITION
Using laser: Chameleon MPX, Version, Serial Number, Device Type: APE GmbH, Chameleon OPO, S05221, V 0.5.3376, SW4.4.19.6
Acquiring with: ScanImage v5.6.1 on MATLAB 9.5.0.944444 (R2018b)
**Laser set to switch off at the end of acquisition
BakingTray will slice the final imaged section off the block**
Setting laser watchdog timer to 2400 seconds
...
"
The lines relating to the "Laser off" and "Slice last" boxes in the preview window are not in the log anymore. 

Best regards, 
Hugo

raacampbell, 2020-12-10:
That is very observant of you!
I removed those log lines because whether or not those events happen can be altered by the user after the acquisition has started. Thus those lines are potentially misleading.

closed 2020-12-10 by raacampbell

HugoB, 2020-12-10:
Is there any way to still keep track of these parameters ?

raacampbell, 2020-12-10:
No there isn't. In reality there never was because if the user modified them after the acquisition started there as no log of this. What are you seeking to do? Maybe there is a better or more direct solution.

HugoB, 2020-12-10:
Ok that's fine, it's not so important. 
Thank you