SWC-Advanced-Microscopy/BakingTray #356: Z stack size mismatch with overlap Z planes not 0

HugoB, 2021-02-09:
Hello Rob, 

I didn't run acquisition for a while now and I am testing the recent changes that you made. 
While running a test acquisition, at the end of the first section, I have the following error : 

"
Unable to perform assignment because the size of the left side is 420-by-420-by-2-by-4 and the size of the right side is
420-by-420-by-4-by-4.

Error in BT/placeNewTilesInPreviewData (line 46)
            obj.lastPreviewImageStack(y,x,:,:) = obj.downSampledTileBuffer;
"

By looking at how lastPreviewImageStack is defined in initialisePreviewImageData, I see that : 
(line 45)
"obj.lastPreviewImageStack = zeros([imRows,imCols, ...
        obj.recipe.mosaic.numOpticalPlanes, ...
        obj.scanner.maxChannelsAvailable],'int16');
" 
Which seems to confirm what the error reports. The number of overlap z planes is not taken into account when lastPreviewImageStack is initialized. 

The new area selector feature is really great by the way ! 

Best, 
Hugo

raacampbell, 2021-02-09:
Righto, will look into this soon.

raacampbell, 2021-02-10:
Fixed: eb4c8468eaa8bee3aa51253884937e66d20b491d
Will merge all this into `master` this afternoon.

closed 2021-02-10 by raacampbell