SWC-Advanced-Microscopy/BakingTray #397: BT.takeRapidPreview error : "Unrecognized function or variable 'trigLine'"

HugoB, 2021-08-04:
I recently switched to SI Basic and this message appears when I try to run a Preview Scan : 
>BT.takeRapidPreview`` failed with error message:
>Unrecognized function or variable 'trigLine'.

The Preview GUI freezes and I have to restart BT because it doesn't want to abort the failed fast preview (nothing happens when I press "Stop > Yes : STOP now")

Have you any idea to fix it ? 

Thank you by advance

raacampbell, 2021-08-06:
Did you check if this issue still happens with the latest version on the `master` branch?

HugoB, 2021-09-02:
Yes, I have the same issue.

raacampbell, 2021-09-02:
Can you try the `prerelease` branch? I just merged changes into it. Can you also confirm which version of Basic you are using? I have verified fully running systems on Basic 2020.1.4 but Basic 2021.0.0 I think indeed I have not tried a preview scan yet. I did sort out some bugs with 2021, however, so there should not be much left to do in order to get it fully working.

HugoB, 2021-09-02:
I think it's the Basic 2020.1.3 
How does one can keep the scanimage version up to date ?

raacampbell, 2021-09-02:
Great! I'd suggest you use `master` because `dev` is unstable and 2020.1.4 definitely works with `master`.
I'll soon fully test 2021, but so far most things seem to work.